Levi--G/IMU-WhoAmIVerifier #1: I'm having this error and its nothing wrong with the soldering

GabrielRosa3252, 2022-03-11:
[image]
 this is the error

[ERR] Found unknown I2C device on address 0x68 with possible id's:
Reg: 0x75 Value:0x72
Reg: 0x00 Value:0xFD

Levi--G, 2022-03-11:
Which chip are you trying to detect? do you have a picture of the pcb?

GabrielRosa3252, 2022-03-11:
[image]

Its mpu6050
I can't take picture of the chip but I've checked all the imus and all of them is mpu6050
I've buyed 4 imus but only one worked and the rest are returning this message

You said I can change the id but how can I do this?

closed 2022-04-26 by Levi--G