Levi--G/IMU-WhoAmIVerifier #2: unknown I2C device Reg: 0x75 Value:0x74 Reg: 0x00 Value:0xC0

JoshuaRNichols, 2022-04-26:
Is it possible that this is an unidentified IMU?

The silkscreen says "MPU-9250/6500/9255"

```
[ERR] Found unknown I2C device on address 0x68 with possible id's:
Reg: 0x75 Value:0x74
Reg: 0x00 Value:0xC0
```

Levi--G, 2022-04-26:
This is the same case as issue #1 Since the id is close to mpu9250 but not the same, either this is some factory sample or special series or it seems to be a knockoff, there is no way to tell for sure unless the ID matches a known id... MPU9250 have been out of production for so long it should be near impossible to find official chips anymore.

JoshuaRNichols, 2022-04-26:
It was sold to me as this: https://www.amazon.com/dp/B09STXTBSZ 

"AITRIP 5PCS MPU-6500 GY-6500 Sensor Module 3 Axis Gyro + 3 Axis Accelerometer Replace MPU6000 MPU6500

- This module contains 16-bit ADC output of 3 axis accelerator and 3 axis gyro
- Model: GY-6500 ; Chip: MPU6500
- Power voltage: 3 - 5V
- Gyro range: +/-250, +/-500, +/-1000, +/-2000 dps
- 5PCS MPU-6500 GY-6500 6-Axis Gyroscope Accelerometer Sensor Module

Brand Name | AITRIP
Ean | 0701715513457
UNSPSC Code | 41111900
UPC | 701715513457
"

Levi--G, 2022-10-12:
Closing, can't fix fakes sadly

closed 2022-10-12 by Levi--G