Levi--G/IMU-WhoAmIVerifier #6: [ERR] Found unknown I2C device on address 0x28 with possible id's: Reg: 0x75 Value:0x00 Reg: 0x00 Value:0x00

uwidev, 2022-07-23:
[ERR] Found unknown I2C device on address 0x68 with possible id's:
Reg: 0x75 Value:0x00
Reg: 0x00 Value:0x00
Please report this on https://github.com/Levi--G/IMU-WhoAmIVerifier/issues with information above. :)
[ERR] Found unknown I2C device on address 0x69 with possible id's:
Reg: 0x75 Value:0x00
Reg: 0x00 Value:0x00
Please report this on https://github.com/Levi--G/IMU-WhoAmIVerifier/issues with information above. :)
[ERR] Found unknown I2C device on address 0x29 with possible id's:
Reg: 0x75 Value:0x00
Reg: 0x00 Value:0x00
Please report this on https://github.com/Levi--G/IMU-WhoAmIVerifier/issues with information above. :)
[ERR] Found unknown I2C device on address 0x28 with possible id's:
Reg: 0x75 Value:0x00
Reg: 0x00 Value:0x00
Please report this on https://github.com/Levi--G/IMU-WhoAmIVerifier/issues with information above. :)

[image]
[image]

MPU-6500s bought from: https://www.aliexpress.com/item/2251832516905563.html?spm=a2g0o.order_list.0.0.21ef1802h6cWvr

Or so they're supposed to be.

uwidev, 2022-07-23:
Potentially resolved. Seems like a bad solder.

closed 2022-07-23 by uwidev

Levi--G, 2022-07-24:
Glad you found it! :)