Levi--G/IMU-WhoAmIVerifier #9: Found unknown I2C device on address 0x0D with possible id's: Reg: 0x75 Value:0xFF Reg: 0x00 Value:0x00

Aistify, 2022-10-12:
Bought a few chips labeled HMC5883L with the chip markings indicating that its actually a QMC 5883L

[ERR] Found unknown I2C device on address 0x0D with possible id's:
Reg: 0x75 Value:0xFF
Reg: 0x00 Value:0x00

Is this still usable?

Levi--G, 2022-10-12:
Have you tried the branch for mags?
https://github.com/Levi--G/IMU-WhoAmIVerifier/tree/FEAT-more-mags
I wasn't able to test it yet but if it works it should at least tell you which it is?

Aistify, 2022-10-12:
[IMU] Found IMU is QMC5883L with device ID 0xFF on (Primary) address 0x0D
[SlimeVR] SlimeVR with this IMU is: NOT SUPPORTED :c

After todays patch it says this now
I guess it works now